emfcamp/Mk3-Hardware #4: Board 2 Issues - Outline, stop, footprints,  misc

mattbrejza, 2016-05-23:
|  Item | Notes  | Progress |
|---|---|---|
| Teardrops on final board, do some test runs on copy of brd file | | |
| Ant pad, less exposed copper and cream | | |
| Round corners on lcd ribbon slot, use 2mm wire as template in milling layer | | |
| Don't like combined SD footprint, needs work | Will fix once decided on final part | |
| Cutout example for screen ribbon | | |
| round corners near 3.3V/0V holes more | | |
| Missing drill hole for screen mounting/location peg | | |
| fiducials in opposite corners | | |
| Larger stop around mcu and fpc connector pads, or make sure board house removes fingers | | |
| change footprints of microusb, switches to final ones | | |
| Remove cream from tag connector |I think just a dpcbs stencil 'artifact' | |
| find wurth electronik passives we can swap in | | |
| 6x6 Tac switch foot print could have smaller pads and a little closer together also add fancy alps pads just in case | | |
| See if we can get a Plastic lanyard clip sample asap and then check the  clip size and what hole it will need. | | |

mattbrejza, 2016-05-23:
Example image of screen peg: https://www.dropbox.com/s/eqn3t87pvbxqiao/IMG_4801.jpg?dl=0

dpslwk, 2016-08-15:
No longer an issue

closed 2016-08-15 by dpslwk